markmal/freeship-plus-in-lazarus #92: Range Check Error exception exporting DXF 3D polyline...

Brian-Drummond, 2022-02-19:
Open example file 730.fbm in Freeship 5.0.30 build 422 .
File/Export/DXF 3D polyline...

> Program exception! 
> Stacktrace:
> Exception class: ERangeError
> Message: Range check error
>   $000000000086BF1E line 1129 of Units/FreeSpline.inc
>   $00000000008763F9 line 767 of Units/FreeSubdivisionControlCurve.inc

Now looking at the source code of Units/Freespline.inc around that area we see
(line 1108)
```
  for J := 1 to NParams do
  begin
    P := Value(Params[J - 1]);
```
...
(line 1127)
```
  for J := 0 to NParams do
  begin
    P := Value(Params[J - 1]);
```
which looks like a range error to me...

Brian-Drummond, 2022-02-19:
Never mind ... I can see it's fixed in Master, I was looking in the source package in "Releases".

Brian-Drummond, 2022-02-25:
Getting the Master to build was another issue ... I finally came to the conclusion that the Gtk3 target **really** wasn't ready for prime time, and I had to convince the Gtk2 target to stop pulling in most of Gtk3. 
This involved rewriting TFileIconGtk.getIconNamesForFileGtk3 (now renamed getIconNamesForFileGtk2) to get file type information by other means than using Gtk3, and deleting the Gtk3 dependencies.

Brian-Drummond, 2022-02-25:
Reopening to remind me to set up a pull request with my stopgap fix for the GTK2 build...

markmal, 2023-03-18:
fixed in version | 5.0.36.490

closed 2024-04-23 by markmal