markmal/freeship-plus-in-lazarus #96: source code for ancillary applications

Cairn23, 2022-11-26:
Where is the source code for the ancillary applications such as propCal, propcal, fppcalc fungleib etc.
Freeship needs these apps in order for itto work fully, I am prepared to do the required work to get them working but need the source code to start with. There are 24 ancillary exe's used within Freeship. was the source contained within the hydronship?

markmal, 2023-01-02:
Hi guys. Unfortunately these apps are proprietary from some scientific institutions of former USSR/Ukraine. To my knowledge the programs written in Fortran for MS/DOS. These called as .exe from the FreeShip.

pafurijaz, 2023-01-02:
> Hi guys. Unfortunately these apps are proprietary from some scientific institutions of former USSR/Ukraine. To my knowledge the programs written in Fortran for MS/DOS. These called as .exe from the FreeShip.

That is sad, I checked in the internet archive but source code links are broken.

Other question @markmal do you think to continue the developing?

And a question about converting to NURBS, can you isolate the code to export subdivision surfaces to NURBS?  I'm asking this because I'd like to somehow use the method FreeShip uses in Blender.  Thank you

markmal, 2023-01-02:
What links did you check?

I did not have much time in 2022. Not sure how much time I'll have in 2023. I can fix simple bugs.
What do you mean by "isolate the code"?

Cairn23, 2023-01-02:
Its good to see that the main code is still alive, but the problem is that when you attempt to use a number of functions which rely on the ancillary applications Freeship crashes. therefore it may be better to either be able to get updated versions of the ancillary apps, or remove the options form the menu's. 
I have also observed some else has raised a similar issue to this in that when calling an ancillary app the main application does not respond well.

markmal, 2023-01-02:
Menu options depend on availability of the ancillary apps (exe files). You can delete the exes that are unstable in your environment. I believe I've tested them in Win10 and Linux Wine they worked.
Ideally it would be better to implement the code natively according to the methods from the original articles. But it is a lot of work and requires subject expertize.

Cairn23, 2023-01-03:
They are not working for me, may be its because I have 64 bit OS, which is more the norm now a days.

markmal, 2023-01-03:
It is still possible to configure 16-bit and 32-bit support in Win10 64-bit.
For example - https://www.groovypost.com/howto/enable-16-bit-application-support-windows-10/

pafurijaz, 2023-01-03:
> What links did you check?

This one internet archive for the source code here in this link https://web.archive.org/web/20161001193944/http://www.hydronship.net/download.htm

> I did not have much time in 2022. Not sure how much time I'll have in 2023. I can fix simple bugs. 

That is a very good news, thanks!

> What do you mean by "isolate the code"?

I tried to figure out several times where the code to export to IGES was, but I couldn't find the code, but if it were possible, I would like to have only the part of the code that converts subdivision surfaces to IGES surfaces, having the code maybe I could find someone to convert the code to python to make a blender add-on.

markmal, 2023-01-03:
1. Additional freeware external modules - are binary executables. Not the sources.
2. What are you trying to achieve with IGES in Blender, what kind of add-on?

pafurijaz, 2023-01-04:
>     2. What are you trying to achieve with IGES in Blender, what kind of add-on?
A simple add-on that allows me to export subdivision surfaces in IGES format like FreeShip does, but directly from Blender without using FreeShip to do the conversion.

Hi Mark, I use Blender to create the models and import the WRL files into FreeShip, here are some models I've made in the past and they've almost all been imported into FreeShip, for some hydrostatics calculations.
here --> my [Blender and FreeShip Models](https://grabcad.com/pa.furijaz-1/models)
[image]

However it would be nice to make a basic version of FreeShip for Blender, then maybe over time it could evolve into something more complete and reliable.

markmal, 2023-01-04:
may be in your case it would be simpler to make a Blender add-on for some hydrostatics calcs?

pafurijaz, 2023-01-04:
> may be in your case it would be simpler to make a Blender add-on for some hydrostatics calcs?

Yes this would surely be useful, but I need to export the models from Blender to Rhinoceros or other CAD, this is what interests me for now.

like these examples all created in Blender and imported from FreeShip's IGES

[image]
[image]
[image]
[image]

closed 2024-04-23 by markmal