markmal/freeship-plus-in-lazarus #101: Scaling background image with mouse

froussel, 2023-10-20:
Good morning,

I very often try to make an fbm file from a shape plan to make the shapes evolve.
I can't use the scale correctly in the background image.
Would it be difficult to additionally add a frame to the image import with a corner at the top and right which would be adjustable with the mouse, allowing scaling as for the origin where the mouse allows move the image?

markmal, 2024-04-18:
Hi Froussel,
I see it is not convenient. I'll think how to improve.

markmal, 2024-04-26:
@froussel I've added "Frame" function to Background Image pop-up menu. It will be soon in release starting v 5.0.37.*
You should see yellow frame. Pull its edges when you see <-> cursor. Exit this mode with right click.

closed 2024-05-08 by markmal