markmal/freeship-plus-in-lazarus #103: Non symmetic layer doesn't work

blackhat81, 2024-02-01:
Hello, i can't perform non symmetric layer feature: i need to deaw a model of a gondola like boats. I've tried precompiled version but i'm not able to perform this option.

pafurijaz, 2024-02-02:
Hi sadly this app seem abandoned ,only a changes at the beginning of 2023, but basically nothing significant, I don't think there will be any other updates, it's been down for years now  since 2020.

This application should be rewritten in another more modern language, perhaps others could contribute.

blackhat81, 2024-02-05:
OK. Found a way to perform non symmetric layer feature. 
It seems work fine.

markmal, 2024-04-18:
@blackhat81 
sounds great!

@pafurijaz 
, yes, I really do not have much time. Busy with work and other stuff.

closed 2024-04-18 by markmal