markmal/freeship-plus-in-lazarus #106: Cannot Develop

LynxFabrication, 2024-03-22:
Every time I try to develop I get the following error;

> Program exception! 
> Stacktrace:
> 
> Exception class: EObjectCheck
> Message: Object reference is Nil
>   $000000000087CBAC  UNROLL,  line 1883 of Units/FreeDevelopedPatch.inc
>   $000000000089F6D4  UNROLL,  line 1340 of Units/FreeSubdivisionLayer.inc
>   $0000000000955FF2  LAYER_DEVELOP,  line 6947 of Units/FreeEdit.inc
>   $000000000053DCAA  DEVELOPLAYERSEXECUTE,  line 2648 of Forms/Main.pas
>   $000000000055EBBC
>   $00000000005CE509  EXECUTE,  line 246 of include/customaction.inc
>   $000000000055E8B2
>   $00000000005C77DA  DOCLICKED,  line 293 of include/menuitem.inc
>   $000000000046EE5A
>   $0000000000757515  DELIVERMESSAGE,  line 3780 of gtk2proc.inc
>   $00000000007EFB0E  GTK2MENUITEMACTIVATE,  line 139 of gtk2wsmenus.pp
>   $000000000089F6D4  UNROLL,  line 1340 of Units/FreeSubdivisionLayer.inc
>   $0000000000955FF2  LAYER_DEVELOP,  line 6947 of Units/FreeEdit.inc
>   $000000000053DCAA  DEVELOPLAYERSEXECUTE,  line 2648 of Forms/Main.pas
>   $000000000055EBBC
>   $00000000005CE509  EXECUTE,  line 246 of include/customaction.inc
>   $000000000055E8B2
>   $00000000005C77DA  DOCLICKED,  line 293 of include/menuitem.inc
>   $000000000046EE5A
>   $0000000000757515  DELIVERMESSAGE,  line 3780 of gtk2proc.inc
>   $00000000007EFB0E  GTK2MENUITEMACTIVATE,  line 139 of gtk2wsmenus.pp

This is my system info;
> System:
>   Kernel: 5.15.0-58-generic x86_64 bits: 64 compiler: gcc v: 11.3.0 Desktop: Xfce 4.18.1
>     tk: Gtk 3.24.33 info: xfce4-panel wm: xfwm 4.18.0 vt: 7 dm: LightDM 1.30.0
>     Distro: Linux Mint 21.3 Virginia base: Ubuntu 22.04 jammy
> Machine:
>   Type: Laptop System: Acer product: Aspire E5-571 v: V1.26 serial: <superuser required>
>   Mobo: Acer model: EA50_HB v: V1.26 serial: <superuser required> UEFI-[Legacy]: Insyde v: 1.26
>     date: 12/18/2014
> CPU:
>   Info: dual core model: Intel Core i5-5200U bits: 64 type: MT MCP smt: enabled arch: Broadwell
>     rev: 4 cache: L1: 128 KiB L2: 512 KiB L3: 3 MiB
>   Speed (MHz): avg: 2546 high: 2678 min/max: 500/2700 cores: 1: 2495 2: 2518 3: 2494 4: 2678
>     bogomips: 17560
>   Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
> Repos:
>   Packages: apt: 2598
>   No active apt repos in: /etc/apt/sources.list
>   Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
>     1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
>   Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
>     1: deb http: //packages.linuxmint.com virginia main upstream import backport
>     2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
>     3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
>     4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
>     5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
>   Active apt repos in: /etc/apt/sources.list.d/ubuntuhandbook1-ppa-jammy.list
>     1: deb [signed-by=/etc/apt/keyrings/ubuntuhandbook1-ppa-jammy.gpg] https: //ppa.launchpadcontent.net/ubuntuhandbook1/ppa/ubuntu jammy main
> Info:
>   Processes: 229 Uptime: 2h 59m wakeups: 2 Memory: 3.75 GiB used: 2.12 GiB (56.5%) Init: systemd
>   v: 249 runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12 Client: Unknown python3.10 client
>   inxi: 3.3.13

LynxFabrication, 2024-03-24:
Update: Replicated same error on Windows 10 Machine.

markmal, 2024-04-18:
Hi @LynxFabrication 

this app is developed in Lazarus IDE 2.0.10. not gcc.

markmal, 2024-05-08:
I found the bug. Fixing....