markmal/freeship-plus-in-lazarus #108: Unable to see my file when navigating with File>Import>Chines

kevonroberts, 2024-05-08:
When I try to import a chines.txt file using "File>Import>Chines ...", I do not see my file after navigating to the folder where it exists. I don't see any *.txt files as a matter of fact.

markmal, 2024-05-08:
What is your OS and FreeShip version?

kevonroberts, 2024-05-08:
OS:
Edition	Windows 11 Home
Version	23H2
Installed on	‎11/‎26/‎2022
OS build	22631.3527
Experience	Windows Feature Experience Pack 1000.22700.1003.0

Freeship Version:

Product version | 5.1.1
-- | --
Program version | 5.1.1.504
Git Revision | 462
Git Commit | 0713a06386b8314480c698156004cf2e76e3b2cd
Build date | 2024/05/06 09:06:18
Compiler version | 3.2.2
Target CPU | x86_64
Target OS | Win64

</body></html>

kevonroberts, 2024-05-26:
Resolved

closed 2024-05-26 by kevonroberts