markmal/freeship-plus-in-lazarus #109: Program exception!

pafurijaz, 2024-06-12:
Hi in the latest betta

I get this error and the application close immediately

Program exception! 
Stacktrace:

Exception class: EObjectCheck
Message: Object reference is Nil
  $00000000009B0C9B  FORMCREATE,  line 296 of Forms/FreeSplashWndw.pas
  $00000000004C9187  DOCREATE,  line 922 of include/customform.inc
  $00000000004C6E27  AFTERCONSTRUCTION,  line 77 of include/customform.inc
  $00000000004CFFD4  CREATE,  line 3213 of include/customform.inc
  $0000000000549A10  SHOWSPLASHWINDOW,  line 1797 of Forms/Main.pas
  $0000000000544CCD  FORMACTIVATE,  line 1059 of Forms/Main.pas
  $00000000004D6A76  NOTIFYACTIVATEHANDLER,  line 739 of include/application.inc
  $00000000004D8A59  ACTIVATE,  line 1440 of include/application.inc
  $00000000004DA559  INTFAPPACTIVATE,  line 1998 of include/application.inc
  $0000000000705F37  SETAPPACTIVE,  line 2728 of gtk2widgetset.inc
  $00000000007FAE7E  GTK2FORMEVENT,  line 294 of gtk2wsforms.pp
  $00007689069434D7
  $00000000004C9187  DOCREATE,  line 922 of include/customform.inc
  $00000000004C6E27  AFTERCONSTRUCTION,  line 77 of include/customform.inc
  $00000000004CFFD4  CREATE,  line 3213 of include/customform.inc
  $0000000000549A10  SHOWSPLASHWINDOW,  line 1797 of Forms/Main.pas
  $0000000000544CCD  FORMACTIVATE,  line 1059 of Forms/Main.pas
  $00000000004D6A76  NOTIFYACTIVATEHANDLER,  line 739 of include/application.inc
  $00000000004D8A59  ACTIVATE,  line 1440 of include/application.inc
  $00000000004DA559  INTFAPPACTIVATE,  line 1998 of include/application.inc
  $0000000000705F37  SETAPPACTIVE,  line 2728 of gtk2widgetset.inc
  $00000000007FAE7E  GTK2FORMEVENT,  line 294 of gtk2wsforms.pp
  $00007689069434D7

markmal, 2024-06-12:
Yes, I've found and fixed the issue. Before new release, as a workaround you can add parameter
--nosplash-I-ACCEPT-GPLv3-TERMS-AND-CONDITIONS

FreeShip.exe --nosplash-I-ACCEPT-GPLv3-TERMS-AND-CONDITIONS

pafurijaz, 2024-06-13:
Thanks, but I reinstalled the previous version, however I use the Linux version.
Greetings

commit a1e70ebb (2024-06-14): Fixed #109 Program exception
closed 2024-06-14 by markmal