markmal/freeship-plus-in-lazarus #110: This is an update request of VRML1 format

pafurijaz, 2024-06-14:
Hi, I wonder if it is possible to update the import format from VMRL1 to VRML2, perhaps if it were not possible to add another format such as OBJ which can be exported from many applications, this would allow meshes to be imported from other programs in a simpler way for anyone.

The VRML1 format is deprecated and it is impossible to find applications that still save in this format.

Greetings.

markmal, 2024-06-17:
Hi, I'll take a look VRML2 and OBJ.

pafurijaz, 2024-06-17:
That is an awesome news, thanks you very much!

markmal, 2024-06-17:
Some examples of VRML2 file(s) would be helpful. Could you send it here?

pafurijaz, 2024-06-17:
Hi, I have attached several formats including x3d which is the successor of VRML2 and also the OBJ format, if you need I will also prepare the files in VRML1 if you want to make comparisons.

[file_sample.zip](https://github.com/user-attachments/files/15877347/file_sample.zip)

for any sample ask I will prepare them.
Greetings

pafurijaz, 2024-06-17:
This are the previous boat file but in VRML1
[HamptonBoat1938_half_VRML1.zip](https://github.com/user-attachments/files/15877450/HamptonBoat1938_half_VRML1.zip)

markmal, 2024-06-17:
Thanks, that is helpful!

pafurijaz, 2024-06-18:
Hi Mark, I was looking at the different formats and in my opinion the first choice remains the OBJ format, and compared with VRML1, they are very similar, I think this may be the most suitable even if it has different functionality, and the second in order is definitely the x3d format. 

This is the cube code in VRML1 format

```
#VRML V1.0 ascii
Separator {
  Coordinate3 {
          point [
-1.000000 1.000000 1.000000,
-1.000000 1.000000 -1.000000,
1.000000 1.000000 1.000000,
1.000000 1.000000 -1.000000,
-1.000000 -1.000000 1.000000,
-1.000000 -1.000000 -1.000000,
1.000000 -1.000000 1.000000,
1.000000 -1.000000 -1.000000

          ]
        }
  IndexedFaceSet {
        coordIndex [
0 4 6 2 -1 
3 2 6 7 -1 7 6 4 5 -1 5 1 3 7 -1 1 0 2 3 -1 
5 4 0 1 -1 
        ]
  }
}

```

and this the one in OBJ

```
# Blender 4.2.0 Beta
# www.blender.org
o Cube
v -1.000000 1.000000 1.000000
v -1.000000 1.000000 -1.000000
v 1.000000 1.000000 1.000000
v 1.000000 1.000000 -1.000000
v -1.000000 -1.000000 1.000000
v -1.000000 -1.000000 -1.000000
v 1.000000 -1.000000 1.000000
v 1.000000 -1.000000 -1.000000
vn -0.0000 -0.0000 1.0000
vn 1.0000 -0.0000 -0.0000
vn -0.0000 -1.0000 -0.0000
vn -0.0000 -0.0000 -1.0000
vn -0.0000 1.0000 -0.0000
vn -1.0000 -0.0000 -0.0000
vt 0.625000 0.500000
vt 0.875000 0.500000
vt 0.875000 0.750000
vt 0.625000 0.750000
vt 0.375000 0.750000
vt 0.625000 1.000000
vt 0.375000 1.000000
vt 0.375000 0.000000
vt 0.625000 0.000000
vt 0.625000 0.250000
vt 0.375000 0.250000
vt 0.125000 0.500000
vt 0.375000 0.500000
vt 0.125000 0.750000
s 0
f 1/1/1 5/2/1 7/3/1 3/4/1
f 4/5/2 3/4/2 7/6/2 8/7/2
f 8/8/3 7/9/3 5/10/3 6/11/3
f 6/12/4 2/13/4 4/5/4 8/14/4
f 2/13/5 1/1/5 3/4/5 4/5/5
f 6/11/6 5/10/6 1/1/6 2/13/6

```

and this is the x3d of the same cube

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE X3D PUBLIC "ISO//Web3D//DTD X3D 3.0//EN" "http://www.web3d.org/specifications/x3d-3.0.dtd">
<X3D version="3.0" profile="Immersive" xmlns:xsd="http://www.w3.org/2001/XMLSchema-instance" xsd:noNamespaceSchemaLocation="http://www.web3d.org/specifications/x3d-3.0.xsd">
	<head>
		<meta name="filename" content="Cube.x3d" />
		<meta name="generator" content="Blender 4.2.0 Beta" />
	</head>
	<Scene>
		<NavigationInfo headlight="true"
		                visibilityLimit="0.0"
		                type='"EXAMINE", "ANY"'
		                avatarSize="0.25, 1.75, 0.75"
		                />
		<Background DEF="WO_World"
		            groundColor="0.051 0.051 0.051"
		            skyColor="0.051 0.051 0.051"
		            />
		<Transform DEF="Cube_TRANSFORM"
		           translation="0.000000 0.000000 0.000000"
		           scale="1.000000 1.000000 1.000000"
		           rotation="0.000000 0.707107 0.707107 3.141593"
		           >
			<Transform DEF="Cube_ifs_TRANSFORM"
			           translation="0.000000 0.000000 0.000000"
			           scale="1.000000 1.000000 1.000000"
			           rotation="0.000000 1.000000 0.000000 0.000000"
			           >
				<Group DEF="group_ME_Cube">
					<Shape>
						<Appearance>
							<Material DEF="MA_Material"
							          diffuseColor="0.800 0.800 0.800"
							          specularColor="0.401 0.401 0.401"
							          emissiveColor="0.000 0.000 0.000"
							          ambientIntensity="0.000"
							          shininess="0.500"
							          transparency="0.0"
							          />
						</Appearance>
						<IndexedFaceSet solid="false"
						                texCoordIndex="0 1 2 3 -1 4 5 6 7 -1 8 9 10 11 -1 12 13 14 15 -1 16 17 18 19 -1 20 21 22 23 -1 "
						                coordIndex="0 4 6 2 -1 3 2 6 7 -1 7 6 4 5 -1 5 1 3 7 -1 1 0 2 3 -1 5 4 0 1 -1 "
						                >
							<Coordinate DEF="coords_ME_Cube"
							            point="1.000000 1.000000 1.000000 1.000000 1.000000 -1.000000 1.000000 -1.000000 1.000000 1.000000 -1.000000 -1.000000 -1.000000 1.000000 1.000000 -1.000000 1.000000 -1.000000 -1.000000 -1.000000 1.000000 -1.000000 -1.000000 -1.000000 "
							            />
							<TextureCoordinate point="0.6250 0.5000 0.8750 0.5000 0.8750 0.7500 0.6250 0.7500 0.3750 0.7500 0.6250 0.7500 0.6250 1.0000 0.3750 1.0000 0.3750 0.0000 0.6250 0.0000 0.6250 0.2500 0.3750 0.2500 0.1250 0.5000 0.3750 0.5000 0.3750 0.7500 0.1250 0.7500 0.3750 0.5000 0.6250 0.5000 0.6250 0.7500 0.3750 0.7500 0.3750 0.2500 0.6250 0.2500 0.6250 0.5000 0.3750 0.5000 " />
						</IndexedFaceSet>
					</Shape>
				</Group>
			</Transform>
		</Transform>
	</Scene>
</X3D>
```

OBJ is a neutral format now used everywhere, simple to read and implement, but above all it also has many modern features.

Greetings

pafurijaz, 2024-06-18:
These are other sample in VRML1 and OBJ
[Sample02_VRML1-OBJ.zip](https://github.com/user-attachments/files/15889424/Sample02_VRML1-OBJ.zip)

markmal, 2024-07-05:
Hi Pafurijaz,
here is FreeShip binary with VRML2 import. 
Please backup the existing FreeShip binary and replace it with this one and test to import VRML2 files.
The program autodetects VRML version.
[FreeShip.zip](https://github.com/user-attachments/files/16113594/FreeShip.zip)

I think next improvement should be import OBJ. It is much simpler then X3D.

pafurijaz, 2024-07-05:
Great I will test tomorrow, and OBJ is the right choice, give the ability to import file made almost with every application.

pafurijaz, 2024-07-07:
Hi I tried but it doesn't seem to work, I still have the old menu I assume you compiled something else, in fact it doesn't open the VRML 2 format

This is what I have as error

`Found "appearance", "}" was expected at 740,4653121`

Greetings

markmal, 2024-07-07:
Hi, which file did you tried to load? Could you upload it here?
The menu is same. Program autodetects the format.

pafurijaz, 2024-07-07:
Sorry, it was my mistake, I took a file that wasn't correct, too many tests in that directory and I must have chosen the wrong one, it works very well, I didn't encounter any problems, congratulations, it's a shame that the only one that allows you to do a file conversion to VRML2 without triangulation is meshconv, all the ones I've tried I add triangulation on the mesh.
[image]

pafurijaz, 2024-07-07:
here the application that I use to make conversion also to VRML1 the only one that do that.

https://www.patrickmin.com/meshconv/

Simple command usage:

VRML 1 
./meshconv -c wrl -vrmlver 1 source.obj

VRML 2
./meshconv -c wrl -vrmlver 2 source.obj

pafurijaz, 2024-07-07:
However this is the file that has a problem, it is triangulated and I have this error, even if ti was a VRML2, and conversion was done via MeshLab, Blender for example import it without issues.
[image]
[Happy_Clam_17ft_meshLab.wrl.zip](https://github.com/user-attachments/files/16119839/Happy_Clam_17ft_meshLab.wrl.zip)

Greetings

markmal, 2024-07-07:
Great. I'll investigate with the file.

pafurijaz, 2024-07-07:
Thanks you very much,  I found an other application that make this conversion nicely and works fine and its gCAD3D, while file exported with FreeCAD has the same issues, of previous one made with MeshLab, attached all in the zip Below.
[Happy_Clam_17ft_FreeCAD_gCAD3D.zip](https://github.com/user-attachments/files/16119857/Happy_Clam_17ft_FreeCAD_gCAD3D.zip)

gCAD3D can be downloaded here, is pretty useful in many cases and very small application and opensource.
https://gcad3d.org/

Greetings

markmal, 2024-07-07:
I see the Shape is inside Transform. That was not expected. I need to add this. 
I will not apply actual scale and translation to coord, assuming it is always  scale 1 1 1
  translation 0 0 0
what is creaseAngle?

pafurijaz, 2024-07-07:
I'm not sure what it is creaseAngle, but I assume it refers to smooth shading, but I believe that part is not supported by FreeShip, and you can omit it and ignore everything about that part.

[image]

markmal, 2024-07-08:
please test this
[FreeShip.zip](https://github.com/user-attachments/files/16132117/FreeShip.zip)

pafurijaz, 2024-07-08:
Nice now the file exported also with MeshLab, works very good, but the FreeCAD file has a new errors, but I think is fine also without further changes, however this is the error that the window reports. 

**Found "DEF", "[" was expected at 5,5**

Thanks for your effort these changes will give many people the ability to use FreeShip with other applications.
Greetings

pafurijaz, 2024-07-08:
Just found an other possible issues with VRML2, often can have materials that can be omissied or at latest in the future added as a FreeShip's layers.

**Found "__defaultMaterialFace", "{" was expected at 8,18**

This file was exported with Mayo a free  open source application that can be downloaded here https://github.com/fougue/mayo/releases

This is the file converted from Obj into VRML2 with Mayo
[Viko_20.zip](https://github.com/user-attachments/files/16134030/Viko_20.zip)

Greetings

pafurijaz, 2024-07-08:
last file, this is the previous viko_20 but simple without material in order to have a file to compare.
[Viko_20_meshcon.zip](https://github.com/user-attachments/files/16134165/Viko_20_meshcon.zip)
:)

pafurijaz, 2024-07-09:
I have done several tests, and it seems that all of them are fine except one done with FreeCAD, which however is not entirely reliable as a program, however the file exported from FreeCAD is opened without problems by Blender and MeshLab, but with CAD Assistant and Mayo, there they are errors, so it could be something not compliant, while now the files saved with CAD Assistant and Mayo are imported without problems, I'd say it could be fine.

Since I don't have any other programs I wanted to do some tests with an online converter (https://fabconvert.com/convert/vrml), this file is not opened by FreeShip but with Blender, FreeCAD, MeshLab there are no problems, while with Mayo and Cad Assistant I can't open it. 
so I don't know how compliant it might be. 
But in any case you find it attached in the Zip together with the FreeCAD one, where the problem seems to be the shape string, which is a material, but FreeShip indicates this error: 

**Exception in VRML Load:"ccw" is an invalid integer**

For me now I would say that this makes the VRML2 import of these test very reliable. 
[FabConvert_FreeCAD_viko_20_VRML2.zip](https://github.com/user-attachments/files/16149696/FabConvert_FreeCAD_viko_20_VRML2.zip)

Greetings

markmal, 2024-07-10:
Hi, they all use some poorly documented features like children field with one value not in [ ... ]
or coordindex without last -1.
I've fixed the above.

pafurijaz, 2024-07-10:
Thanks for what you are doing, I saw the VRML 2 spec, and they are really complicated, which I found in this link. https://tecfa.unige.ch/guides/vrml/vrml2/spec/index.html

However, Wings 3D is one of the few programs that allows you to preserve OBJ files with the square topology, which is essential for having correct surfaces in FreeShip. 
Wings 3D is an application that could act as a bridge for everyone, and is still being developed today, and it's the only one that creates files that are imported by all the programs I've tested, but unfortunately in FreeShip they all have a problem with the same line of code which is the **material**.

Errors with Wings 3D VRML 2

Cube: `Found "Material", "{" was expected at 6,22`
Suzanne: `Found "default", "{" was expected at 6,22`
Sphere: `Found "default", "{" was expected at 6,22`
Torus : `Found "default", "{" was expected at 6,22`

Wings 3D is also a simple modeler that many could use to create files for FreeShip, and could thus allow many to use it to make models to analyze and then develop in FreeShip, I hope you can find the time and desire to do this fix.

The Wavefront OBJ format is much simpler than VRML 2 and still extremely widespread today and also used by modern 3D and CAD programs, I found the specifications here in this link [Wavefront **.obj** Specification ](https://paulbourke.net/dataformats/obj/) and the material at this link [Wavefront **.mtl** Specification ](https://www.fileformat.info/format/material/index.htm)

I will attach a zip with VRML2 and OBJ samples

[Wing3D_VRML2_and_Blender_OBJ.zip](https://github.com/user-attachments/files/16164579/Wing3D_VRML2_and_Blender_OBJ.zip)

Greetings.

markmal, 2024-07-11:
The issue "Found "Material", "{" was expected at 6,22" is fixed.
However Viko_20_Wings3d_01.wrl does not look correct above the hull. The hull looks Ok.
Please confirm it looks same loaded in Wings3D.
[FreeShip.zip](https://github.com/user-attachments/files/16177816/FreeShip.zip)

markmal, 2024-07-11:
Yes, I opened it in wings3d and in view3dscene. It looks good. Something is wrong with FreeShip.
looking ...

pafurijaz, 2024-07-11:
I will test, soon as I can, but I think you have already made a great changes that will help a lot many users that are straggling with boat modeling with FreeShip.
Greetings

pafurijaz, 2024-07-11:
Wow now works fine with file exported with Wings 3D
[image]
[image]

About Viko maybe it's a problem related with File the vertex order is possible that have some issues because I used an old Blender Version to make, but recentley the Core mesh was updated.
This new attached seem working fine, very great.
[image]
[Viko_20_V_new.zip](https://github.com/user-attachments/files/16179587/Viko_20_V_new.zip)

Greetings

pafurijaz, 2024-07-11:
Works very Good, after that I will do a brief document to export file correctly with Wings 3D and Blender for FreeShip. :)
[image]

markmal, 2024-07-11:
Great! Thanks! So I do not need to investigate?

pafurijaz, 2024-07-11:
For me you can avoid in that issues, I think we made many test. I think you can go forward.
Thanks.

markmal, 2024-07-12:
Thanks for the testing and examples!
I'll close the case.

closed 2024-07-12 by markmal