markmal/freeship-plus-in-lazarus #114: Problem with function "Insert Plane"

Kees-Muller, 2024-10-06:
Hello Mark,

Thank you for providing the download-link for Freeship.

I am learning the program, but I have a problem. I wish to use the "Insert Plane Funtion" to add a number of controlpoints. Ik can open the dialog box and select "Transverse" but I cannot change the values. In my case ther is a box with the min value and the max value. But I cant change these numbers. 
Is this a bug in the program or am I doing somthing wrong?

Thank you for your help. 

[image]

markmal, 2024-10-23:
Hi,

please read documentation:

6.5 Insert plane.
This operation intersects all edges that are visible with a plane. If such an intersection point exists,
it will be inserted on the edge. After this, faces that have multiple newly inserted points will be split
by inserting a new edge. This is a convenient way to insert for example a while range of points at a
certain ordinate location. There is also an option to add a controlcurve to the newly created edges.
The type of plane (vertical, horizontal or transverse) can be specified as well as the location by
entering the desired distance into the dialog that appears.

Fields Min and Max value are for information only. These show dimensions of your model in according aspect. The value of Distance field should be between Min and Max.

So this is not a bug.

Kees-Muller, 2024-10-23:
Thank you for your help. I really like the software👍