markmal/freeship-plus-in-lazarus #116: Program crashes when I try to Develop Plates

ChrisBennett65, 2024-12-06:
Hi there. I have created a plan with 3 chines and a flat rockered bottom plus transom. I was initially able to develop and print plates even after I created the transom, but after tweaking a few control points I 
[image]
now get what looks like a null pointer (stack trace below). I am unable to attach the ftm file that is causing this but can send it via email. I attach a jpg to give an idea of what the boat looks like in case that is any help.

Program exception! 
Stacktrace:

Exception class: EObjectCheck
Message: Object reference is Nil
  $0000000100276533 line 1900 of Units/FreeDevelopedPatch.inc
  $000000010028F161 line 1294 of Units/FreeSubdivisionLayer.inc
  $0000000100337124 line 6813 of Units/FreeEdit.inc
  $0000000100094608 line 2625 of Forms/Main.pas
  $00000001000A445C
  $00000001000ED824 line 98 of include/containedaction.inc
  $00000001000EF504 line 246 of include/customaction.inc
  $00000001000A41CB
  $00000001000E9034 line 86 of include/menuitem.inc
  $00000001000E9808 line 293 of include/menuitem.inc
  $0000000100010035
  $0000000100198647 line 2059 of win32/win32callback.inc
  $0000000100199FCC line 2677 of win32/win32callback.inc
  $000000010024AAEE line 387 of win32/win32wsforms.pp
  $00007FFF7F5A23FD
  $00007FFF7F5A249D
  $000000010019B18D line 407 of win32/win32object.inc
  $0000000100641548
  $0000000000000000 of (Unknown file)
  $000000010066D2D8
  $0000000000000000 of (Unknown file)
  $0000000000000000 of (Unknown file)
  $0000000000000000 of (Unknown file)
  $0000000002E98590
  $0000000000000000 of (Unknown file)
  $0000000000000000 of (Unknown file)
  $0000000000000000 of (Unknown file)
  $0000000100000000
  $000000010066CBE0
  $0000000000000000 of (Unknown file)
  $000000010008BC00 line 790 of Forms/Main.pas
  $00000000010EE520
  $0000000000000000 of (Unknown file)

markmal, 2025-01-22:
Hi, why you cannot attach the file? It would be very helpful.

ChrisBennett65, 2025-01-22:
Hi Mark

I think the issue may have been already fixed in a later version. I
upgraded to a later version than what I first installed and have not had
that particular issue. I have had the odd crash but these are very
infrequent and can be dealt with by making manual backups. Thanks for your
work in making a very useful product!

Chris

markmal, 2025-01-26:
Thanks Chris!

closed 2025-01-26 by markmal